wpilibsuite/shuffleboard #460: Camera Views

Team3840, 2018-04-04:
Is it possible to add a property to rotate the camera image view in the camera stream input to the widget? As a CSA's we get ask this question frequently to be able to camera view rotation 180 or 90 from the hardware mounted position via software view.

[image]

Team3840, 2018-06-04:
Excellence!  Is there a way we can test it?

SamCarlberg, 2018-06-04:
You can clone the repository with

```
$ git clone https://github.com/wpilibsuite/shuffleboard
```

And check out the pull request with

```command-line
$ git fetch origin pull/478/head:rotate-camera
$ git checkout rotate-camera
```

PR #478 (2018-06-07): Add rotation to camera widgets
Closes #460 

Cameras can be rotated in increments of 90 degrees (0, 90, 180, 270).
closed 2018-06-07 by SamCarlberg