wpilibsuite/shuffleboard #485: Editable Fields in Shuffleboard

dakaleyt, 2018-06-14:
Currently in Shuffleboard, fields listed under the SmartDashboard panel are unable to be edited. This presents a problem to teams that use the getNumber function in order to manage various aspects of their code (e.g. Drive Train Sensitivity pulling a number from the Dashboard).

Suggestions:
a) Make the fields under the SmartDashboard pane editable, similar to the SmartDashboard fields.

SamCarlberg, 2018-06-14:
Can you upload a screenshot?

dakaleyt, 2018-06-14:
I don't have a roboRIO on me but when we were testing code we were unable to edit any fields/numbers.

SamCarlberg, 2018-06-14:
Well, a field is not editable if there's no connection to the robot, or if the entry it modifies no longer exists. Generally, `SmartDashboard.getNumber("key")` and friends doesn't do anything on Shuffleboard unless you've first called `SmartDashboard.setNumber("key", ...)`. https://github.com/wpilibsuite/allwpilib/pull/1022 will fix that for Shuffleboard

SamCarlberg, 2018-11-09:
Closed by wpilibsuite/allwpilib#1022. Use the Shuffleboard API in your robot program to add such fields to the dashboard. You'll also have greater flexibility in where the fields will be placed and how they will be displayed than just text boxes in the Smartdashboard tab. See the pull request for more details.

closed 2018-11-09 by SamCarlberg