wpilibsuite/shuffleboard #495: Splashscreen has an ellipsis truncating the title

calcmogul, 2018-07-28:
**Describe the bug**  
The splashscreen isn't wide enough to display the title, so it gets truncated with an ellipsis.

**To Reproduce**  
Steps to reproduce the behavior:
1. Check out commit 1b916e79 (latest master as of 7/28/18)
2. `./gradlew :app:run`

**Expected behavior**  
The splashscreen should display the whole title.

**Screenshots**  
[image]

Version v1.3.1-34-g1b916e7 built on July 28, 2018
JRE: 1.8.0_172-b11 -- Oracle Corporation
Operating system: Linux 4.17.9-1-ARCH amd64 (unknown distribution)

**Additional context**  
N/A

SamCarlberg, 2018-07-28:
What's your screen resolution?

calcmogul, 2018-07-28:
1920x1080

SamCarlberg, 2018-08-16:
@calcmogul Can you apply this patch and see if the truncation remains? [preloader_patch.txt](https://github.com/wpilibsuite/shuffleboard/files/2294260/preloader_patch.txt)

calcmogul, 2018-08-16:
That fixes it.

PR #504 (2018-08-21): Fix occasional truncation of title in the preloader
Closes #495
closed 2018-08-21 by SamCarlberg