wpilibsuite/shuffleboard #527: Gyro Widget fail

Crossle86, 2018-10-09:
On v1.3.1 when you right click on a numeric item in sources, you are not given the option to show the item with the Gyro widget. If I create a Gryo from the widgets list, nothing happens when I click on Edit Properties. I can't link the  widget to the network tables variable containing my heading.

Note: Shuffleboard would not update itself to 1.3.1, auto update appears to download and backup the current jar but then it exits and you do not have an updated jar in tools, the old jar remains. So in order to test with 1.3.1 I had to clone the project and compile it myself. This does not seem desireable. The release package on github should include the finished jar file.

I'm evaluating shuffleboard for use in 2019. Is shuffleboard being maintained?

SamCarlberg, 2018-10-09:
1. Gyro widgets have never supported raw numbers. If you want to display a heading from a gyro, add the gyro to the dashboard with `SmartDashboard.putData("Gyro", yourGyroObject);`. In 2019, this can be done with `Shuffleboard.getTab("Tab Name").add("Gyro", yourGyroObject);`
2. The update mechanism has been removed from future releases (#514)
3. Yes, shuffleboard is maintained
4. Shuffleboard releases are available from the WPILib IDE plugins (Eclipse for 2018, vscode for 2019 and beyond). Unless you're on a beta team, use the version bundled with the latest Eclipse plugins.

ThadHouse, 2018-10-09:
Actually, 1.3.1 was not in the latest eclipse release. I just pushed the 1.3.1 jar to the 1.3.1 tag on the releases page. 

Note 1.3.1 will require Java 8, and compiling/running from master from this point on will require Java 11.

Crossle86, 2018-10-09:
1. Yes, I finally figured this out after some hours of digging. Why is this not documented? How are people supposed to find this out (other than searching Chief Delphi for clues)?
4. I tried the shipped version and ran into several problems rendering SB unusable. Since that version offered an update to 1.3.1, I assumed that to be the latest release (as shown on github as well) and so I cloned and compiled it since no jar was present at that time. It works much better than the shipped version...

So Im a bit confused...Sam seems to suggest 1.3.1 is not what I  should be using but Thad seems to say it is current. Please clarify.

I asked about maintenance since v1.3.1 was posted march 20...

ThadHouse, 2018-10-09:
1.3.1 is what you should be using. We forgot we didn't do an eclipse update that included that version.

We normally don't post updates for any tools during the offseason. Releases usually only come during the build and competition seasons. We have been actively working on it, just not making releases.

Crossle86, 2018-10-10:
There is a lot going on in SB that I  don't yet understand. It is best to ask questions here or in a post on Chief Delphi?

Crossle86, 2018-10-10:
So the 1.3.1 I compiled myself had issues with the  gyro widget so I downloaded the 1.3.1 jar you pushed  to the release artifacts. When I try to run that jar I get the error "could not find or load main class".

ThadHouse, 2018-10-10:
The first 1.31 I posted immediately when I responded about posting the jar was not the right one. Is your jar ~60mb? If not, redownload it.

Crossle86, 2018-10-10:
There is still an issue with the 1.3.1 code as I compiled it: the Gyro widget does nothing when the edit properties menu choice is clicked.

Crossle86, 2018-10-10:
I downloaded the jar just before reporting the problem. It is 55.6mb same as shown on the release and as I just downloaded it again.

ThadHouse, 2018-10-10:
Oh, are you trying to run on Java 11? I get that error if I attempt to run the release on Java 11. It only supports Java 8. The builds from master are with Java 11.

Crossle86, 2018-10-10:
Ah. Right. So I'd like to continue with the current Java 11 version. Any thoughts on not being able to edit the Gyro widget properties? If that is not something you want to address at this time I will install J8 and run with the released 1.3.1.

SamCarlberg, 2018-10-10:
If a widget has no title, you won't be able to edit its properties (this is on `master`, I think 1.3.1 should be fine). That problem is fixed in #530 

If you want to use the development versions on Java 11, you'll have to build and run from source.

Crossle86, 2018-10-12:
So I updated my 1.3.1 to match #530 and I am able to edit a new  Gyro widget. I added the appropriate sendable as  you indicated earlier in these comments. What I am seeing now is this:
If I have SB auto add to a tab, my Gyro sendable causes a Gyro widget to be created and that widget is linked to the sendable and the Gyro needle tracks the sendable's updates as desired. However, if I select a Gyro widget from the widget list and drop it onto a tab, that widget is not linked to the sendable and does not track changes published by the sendable (which I can see happening in the  network tables viewer). The sendable is  the  same (key name "Gyro", datatype="Gyro") in both cases.

SamCarlberg, 2018-10-12:
Widgets created from the gallery are not backed by any data. You can drag in a gyro widget from the gallery, then drag the gyro source from the NetworkTables source tree onto the widget to set its data source.