wpilibsuite/shuffleboard #539: SHUFFLEBOARD

arshacker, 2018-11-05:
**Describe the bug**  
A clear and concise description of what the bug is.

**To Reproduce**  
Steps to reproduce the behavior:
1. ...
2. ...

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots**  
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Java version [e.g. 1.8.0_171]
 - Version [e.g. 1.3.1]

**Additional context**  
Add any other context about the problem here.

arshacker, 2018-11-05:
After loading a custom plugin I closed shuffleboard. When I try to open it again, I see the title screen saying, " Loading Camera Server plugin". Then the title screen disappears. and nothing else happens.

How can I get a working shuffleboard?

arshacker, 2018-11-05:
Re my shuffleboard problem, Windows, java 1.8.0_181
  After loading a custom plugin I closed shuffleboard. When I try to open it again, I see the title screen saying, " Loading Camera Server plugin". Then the title screen disappears. and nothing else happens.
How can I get a working shuffleboard?

SamCarlberg, 2018-11-05:
Check the most recent log files in the Shuffleboard directory in your home directory. The plugin is probably crashing when it's loaded

AustinShalit, 2018-11-19:
@arshacker We are going to close this PR for now.  If you have more information, please reopen.

closed 2018-11-19 by AustinShalit