wpilibsuite/shuffleboard #567: Graphing values does not have consistent timestep

asid61, 2018-12-31:
[image]

^ picture of the issue. Shuffleboard randomly decides to stop graphing values for some time, then graphs a whole bunch at once. This behavior occurs whenever we are graphing encoder position, error, etc. on Shuffleboard in any timed loop (such as TeleopPeriodic).

SamCarlberg, 2018-12-31:
What version of Shuffleboard do you experience this behavior with?

Since Shuffleboard graphs data as it arrives, having many data points bunched together all at once implies a possible network issue. Does this problem happen when connected over wifi or ethernet?

asid61, 2018-12-31:
Version 1.3.1.
We used WiFi but we know it's not the Wi-Fi's fault, as switching to Smart Dashboard immediately resolves the problem. We experienced issues on several occasions, and it was always fixed by switching to Smart Dashboard.

carbotaniuman, 2020-08-24:
@PeterJohnson Disposition to close.

closed 2020-08-24 by PeterJohnson