wpilibsuite/shuffleboard #573: Unable to launch Shuffleboard from desktop VBS

pizzafox, 2019-01-10:
**Describe the bug**  
Popup error message appears (see below).

**To Reproduce**  
Steps to reproduce the behavior:
1. Install Shuffleboard using [these instructions](https://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/999999-installing-c-and-java-development-tools-for-frc#offline-installer)
2. Run Shuffleboard from desktop

**Expected behavior**  
Shuffleboard to work.

**Screenshots**  

[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Java version: `11.0.1`
 - Version: `2019.1.1`

ThadHouse, 2019-01-10:
Are you running an N version of Windows? That's what we've seen cause that error. If so, follow the instructions about 34 a way down this page to fix that.

https://wpilib.screenstepslive.com/s/currentCS/m/getting_started/l/1028964-known-issues

pizzafox, 2019-01-10:
It turns out we are using an N version of Windows. Installing the [update from Microsoft](https://www.microsoft.com/en-us/software-download/mediafeaturepack) fixed it.

closed 2019-01-10 by pizzafox