wpilibsuite/shuffleboard #576: Camera widget does not initialize stream correctly

prensing, 2019-01-13:
**Describe the bug**  
When I start up SB with a camera widget, it does not use the correct URL, specifically it does not set the compression parameter, despite that being saved in the JSON configuration. Even changing the slider and hitting "Apply" does not do it. I have to change the FPS and/or size, and then it changes. After that, I can change FPS and size back to -1 and the compression clearly sticks (image is appropriately blocky and bandwidth is way down).

**To Reproduce**  
Steps to reproduce the behavior:
1. Set the compression.
2. Save the config and quit.
3. Restart
--> compression is clearly back to the "default" of 80 (I think).

**Expected behavior**  
The compression should be set as in the saved settings. Also, if I change the slider and hit Apply, it should get set.

**Desktop (please complete the following information):**
 - OS: Linux
 - Java version 11.0.1
 - Version 2019.1.1

**Additional context**

SamCarlberg, 2019-01-13:
This is a camera widget in a save file, correct?

prensing, 2019-01-13:
Yes. The tab has just the camera widget.

SamCarlberg, 2019-01-13:
Okay. That problem was fixed in #572 and will be available in the next release. You'll need to save and re-open the file to fix it.

closed 2019-01-13 by SamCarlberg

prensing, 2019-03-14:
This is not actually fixed. I tried 2019.4.1 on Windows (just downloaded the build), and neither compression nor FPS are set when SB starts up. It only works if the image sizes are set.

prensing, 2019-03-14:
Please reopen this issue. Thanks.