wpilibsuite/shuffleboard #578: Could not load the last save file

D0048, 2019-01-17:
**Describe the bug**  
Unable to load configuration files.

**Error Message & To Reproduce**  
Backtraces and my configuration file could be found [here](https://gist.github.com/D0048/9498ab263f0fc5a1fd840a462129b1d5)(Stack trace in second file, please scroll down...)
Hilight:
```
Uncaught exception on JavaFX Application Thread
java.lang.ClassCastException: class com.google.gson.internal.LazilyParsedNumber cannot be cast to class java.lang.Integer (com.google.gson.internal.LazilyParsedNumber is in unnamed module of loader 'app'; java.lang.Integer is in module java.base of loader 'bootstrap')
        at edu.wpi.first.shuffleboard.plugin.cameraserver.widget.CameraServerWidget.applySettings(CameraServerWidget.java:164)
        at edu.wpi.first.shuffleboard.plugin.cameraserver.widget.CameraServerWidget.lambda$initialize$6(CameraServerWidget.java:125)
        at com.sun.javafx.binding.ExpressionHelper$Generic.fireValueChangedEvent(ExpressionHelper.java:360)
        at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:80)
```

**Expected behavior**  
Making my config file load.

**Desktop:**
 - OS: Linux KDE neon User Edition 5.14
 - Java version openjdk version "11.0.1" 2018-10-16
 - Version [2019.1.1]

D0048, 2019-01-17:
I found that even after I remove the camera server causing these exceptions, I am still unable to load the config file.
Here's the terminal output:
```
信息: OpenCV version: 3.4.4
1月 17, 2019 9:06:54 下午 edu.wpi.first.shuffleboard.app.plugin.PluginLoader load
信息: Loaded plugin edu.wpi.first.shuffleboard:CameraServer in 129ms
1月 17, 2019 9:06:54 下午 edu.wpi.first.shuffleboard.app.MainWindowController initialize
信息: Setting up plugins in the UI
1月 17, 2019 9:06:54 下午 edu.wpi.first.shuffleboard.app.Shuffleboard init
信息: Took 215ms to load the main FXML
NT: connect() to 10.71.46.2 port 1735 timed out
1月 17, 2019 9:06:56 下午 edu.wpi.first.shuffleboard.app.Shuffleboard start
信息: Took 2388ms to start Shuffleboard
1月 17, 2019 9:06:56 下午 edu.wpi.first.shuffleboard.app.json.WidgetSaver deserialize
警告: Saved source type is not present, adding destroyed source(s) instead
1月 17, 2019 9:06:56 下午 edu.wpi.first.shuffleboard.app.json.WidgetSaver deserialize
警告: Saved source type is not present, adding destroyed source(s) instead
1月 17, 2019 9:06:56 下午 edu.wpi.first.shuffleboard.app.Shuffleboard lambda$start$3
警告: Could not load the last save file
```

SamCarlberg, 2019-01-22:
This is probably the problem fixed in #572. Update shuffleboard to 2019.2.1 and the problem should be fixed

SamCarlberg, 2019-01-27:
This problem was fixed by PR 572. Updating to version 2019.2.1 will fix the issue.

closed 2019-01-27 by SamCarlberg