wpilibsuite/shuffleboard #582: Add ability to add camera servers based on URL or networktables name.

Daltz333, 2019-02-07:
**Is your feature request related to a problem? Please describe.**  
Unable to add camera server to layout using the FRC Raspberry Pi image.
You currently cannot do this via the Shuffleboard Python API on the Raspberry Pi because you **cannot** install pyfrc on a raspberry pi. A simple solution to this would be to allow the Shuffleboard API to take a NetworkTables entry instead.

**Describe the solution you'd like**  
Allow the Shuffleboard addCamera API to take a NetworkTables entry of the camera server.

**Describe alternatives you've considered**  
Possibly the ability to get PyFRC to build and install on the RaspberryPi? But would having multiple locations setting layouts really be the best idea? I'm open to ideas.

virtuald, 2019-02-10:
Strictly speaking, there's nothing preventing you from installing pyfrc on an RPi -- but that isn't the actual problem anyways, as pyfrc is a helper library that allows you to do various things -- it doesn't have a shuffleboard API. The actual work is all in the wpilib package.

What you really want is the ability to use the `wpilib.shuffleboard` package. If you install the `wpilib` and `robotpy-hal-sim` packages on the RPi via pip, (I haven't tried it, but...) you should be able to use `wpilib.shuffleboard` as expected.

It would be nice if you could just install the `wpilib` package, but unfortunately we do a lot of default imports and so the hal package is also necessary.

Daltz333, 2019-02-10:
That will probably work in the meantime, but my feature request still stands. Thanks @virtuald ❤️

auscompgeek, 2019-02-11:
One can already programmatically add a camera widget by name to Shuffleboard. This is used by the WPILib Shuffleboard API to add a cscore VideoSource to a ShuffleboardContainer.

auscompgeek, 2019-02-11:
If you're trying to ask to be able to feed the WPILib Shuffleboard API a CameraServer name, wpilibsuite/allwpilib#1587 is already open for that.

Daltz333, 2019-02-11:
@auscompgeek I was looking for a method that adds by NetworkTables name/URL. Not the VideoSource as that means Shuffleboard can NOT create a camera widget from a remote stream

auscompgeek, 2019-02-11:
For adding a stream by URL, that is already possible by adding a VideoSource to Shuffleboard. (See the example provided in the issue linked above.)

closed 2022-01-01 by PeterJohnson