wpilibsuite/shuffleboard #583: TAB preferences not saved

gearsincorg, 2019-02-09:
I'm having trouble turning OFF the auto-populate feature on any tabs.

If I use the main TABS menu and  go to preferences, and change them (eg: turn off auto-populate) and save the layout, and also save on exit, Autopopulate is still turned on when I return.

Is there a proper way to turn off auto populate on the SmartDashboard tab?

SamCarlberg, 2019-02-11:
What version of shuffleboard are you using?

gearsincorg, 2019-02-11:
When I went to the help page to find out....It says:  

Version 2019.1.1 built on Feburary 10, 2019..... which seems unlikely since it always shows the current date.

I'm running windows 7

I could not find any instructions on updating.

SamCarlberg, 2019-02-11:
You should be able to update by running the command `WPILib: Install tools from GradleRIO` in the VSCode command palette.