wpilibsuite/shuffleboard #596: Toggle buttons grey out after driver station sleep cycle, even after robot reconnects

20nelson, 2019-03-11:
**Describe the bug**  
On our robot's dashboard, we have three toggle buttons for some manual overrides. These buttons work great, but after closing our driver station laptop and opening it again for the next match, the buttons are greyed out and unable to be toggled, even though the robot is connected. Controls other than toggle buttons show fine, like the camera stream, and some boolean indicators. The booleans are constantly being set in robotPeriodic (I have reasons), so it would make sense that the values would be available whenever the robot is connected. However, they do not show up in the NetworkTables tree even though other data types sent to NT in the same method do. I have verified that network table put calls have correct syntax. 

**To Reproduce**  
Steps to reproduce the behavior:
1. Open shuffleboard with a toggle button on a value being updated in robotPeriodic()
2. Disconnect the robot
2. Put the computer to sleep
3. Wake up the computer and log in
(This can be reproduced without sleeping, but only sometimes. Sleeping the computer makes it happen 100% of the time)
4. Reconnect the robot
5. Toggle buttons are now uneditable

**Expected behavior**  
When connected to the robot, booleans updated in robotPeriodic() should show in the NetworkTables tree and enable controls on the dashboard.

**Screenshots**  
Robot is in the bag currently, so I can't get screenshots. However, the buttons appear just like they do when the robot is disconnected: in the disabled state.

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Java version 11.0.2+7 -- Oracle Corporation
 - Version 2019.4.1

**Additional context**  
We have found a temporary workaround of restarting the dashboard when the robot reconnects to the driver station, but it takes a while to do, inconvenient for field use.

I also experienced this issue in the 2018 off-season, with similar circumstances.

trevnels, 2019-12-27:
Fixed from moving from `SmartDashboard.put` to Shuffleboard API calls

sciencewhiz, 2019-12-27:
Thanks for the PR. However the issue shouldn't be closed until the PR is merged.

trevnels, 2020-01-02:
I don't think this issue is with Shuffleboard itself, but instead with the SmartDashboard API in wpilib not correctly publishing values. Closing this issue and will reopen in allwpilib if I see this behavior again.

closed 2020-01-02 by trevnels