wpilibsuite/shuffleboard #603: Add support for multiple windows

andrewda, 2019-03-30:
**Describe the solution you'd like**  
We have a small secondary monitor on our OI (with a camera stream at the driver's eye level), and want to use a second Shuffleboard window to display the stream.

**Describe alternatives you've considered**  
We are currently using Shuffleboard on our main laptop screen, and displaying the camera stream on a SmartDashboard instance on the secondary monitor. This if fine, besides it not being as elegant and easy to use as Shuffleboard.

SamCarlberg, 2019-04-02:
Seems similar to #135 in that both would benefit from adding drag-and-drop support to tabs

MrRSquared, 2020-01-26:
I second this.  We would love to be able to use a second screen to display a multi-windowed shuffleboard layout.

agasser, 2022-09-27:
Would it be possible to simply allow multiple instances of Shuffleboard to run at once, as a workaround. Maybe a command-line argument to override the default behavior that prevents multiple instances.

PeterJohnson, 2022-09-27:
Potentially, that will need a bit of care though given the global nature of the default save file and similar.

maxlingenfelter, 2024-03-20:
Definitely would love to see this implmented

MrRSquared, 2024-03-21:
In the meantime, a workaround you could try is running an instance of a different driver station interface like one of the html driver stations or glass and shuffleboard at the same time. We have played around with this, but have not explored it in depth or used it at competition.