wpilibsuite/shuffleboard #616: Add option to programatically position widgets within layouts

Oblarg, 2019-06-05:
Currently there is no way to programatically position widgets within a layout.

This could be accomplished without changing the current API; lists could be ordered with the specified widget column index, and grids with the row index/column index.

SamCarlberg, 2019-06-06:
Grid layouts already support this behavior (#604)

The proposal for working with list layouts seems funky - the method names wouldn't correspond to the actual behavior

Oblarg, 2019-06-06:
Any reason the grid layouts use withSize to do this instead of withPosition?  It seems a bit counter-intuitive, and requires software that does this automatically to detect whether a widget is in a tab or a layout to position it correctly.

The list integration might be minorly funky, but I still think it's pretty clear how it behaves; after all, you can think of a list as a 1-by-n grid, so the funkiness is just that only the row index is used and the column index is ignored.