wpilibsuite/shuffleboard #628: Bottom bar is huge and non-resizable

prateekma, 2019-10-18:
**Describe the bug**  
The bottom bar of Shuffleboard with the "NetworkTables: Connected" and play/pause buttons is massive and cannot be resized.

**To Reproduce**  
Steps to reproduce the behavior:
1. Open Shuffleboard 2020 Beta 1

**Expected behavior**  
A clear and concise description of what you expected to happen.

The bar should be way smaller and resizable.

**Screenshots**  

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Java version [e.g. 11.0.4]
 - Version [e.g. 2020.1.1-beta-1]

**Additional context**  
None

AustinShalit, 2019-10-19:
Bisecting shows that the first bad commit is: https://github.com/wpilibsuite/shuffleboard/commit/504fd947e14d114e9b2bc19fd06dcc846fff4d37

SamCarlberg, 2019-10-24:
This typically happens when a widget fails to load and breaks the library view. Can you attach the log file?

@prateekma

ThadHouse, 2019-10-24:
@SamCarlberg we figured out it was opencv not loading correctly. #609 fixes it.

ThadHouse, 2019-10-26:
Closed by #609

closed 2019-10-26 by ThadHouse