wpilibsuite/shuffleboard #629: Shuffleboard fails to load opencv

sciencewhiz, 2019-10-19:
**Describe the bug**  
The shuffleboard camera server plugin fails to load opencv. See log below.

**To Reproduce**  
start shuffleboard

**Expected behavior**  
shuffleboard should load opencv.

**Screenshots**  
```
INFO: Loading plugin edu.wpi.first.shuffleboard:CameraServer:2.0.3
Oct 19, 2019 8:14:30 AM edu.wpi.first.shuffleboard.plugin.cameraserver.CameraServerPlugin onLoad
INFO: OpenCV version: 3.4.7
Oct 19, 2019 8:14:30 AM edu.wpi.first.shuffleboard.plugin.cameraserver.CameraServerPlugin onLoad
SEVERE: Failed to load CV Libraries
java.io.IOException: opencv_java347 could not be loaded from path or an embedded resource.
        attempted to load for platform /windows/x86-64/
        at edu.wpi.first.wpiutil.RuntimeLoader.loadLibrary(RuntimeLoader.java:78)
        at edu.wpi.cscore.CameraServerCvJNI.forceLoad(CameraServerCvJNI.java:58)
        at edu.wpi.first.shuffleboard.plugin.cameraserver.CameraServerPlugin.onLoad(CameraServerPlugin.java:55)
        at edu.wpi.first.shuffleboard.app.plugin.PluginLoader.load(PluginLoader.java:303)
        at edu.wpi.first.shuffleboard.app.Shuffleboard.init(Shuffleboard.java:84)
        at com.sun.javafx.application.LauncherImpl.launchApplication1(LauncherImpl.java:824)
        at com.sun.javafx.application.LauncherImpl.lambda$launchApplication$2(LauncherImpl.java:195)
        at java.base/java.lang.Thread.run(Thread.java:834)
```

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Java version 11.04.11
 - Version 2020.1.1-beta-1

AustinShalit, 2019-10-19:
This does not work on macOS either:
```
SEVERE: Failed to load CV Libraries
java.io.IOException: opencv_java347 could not be loaded from path or an embedded resource.
        attempted to load for platform /osx/x86-64/

        at edu.wpi.first.wpiutil.RuntimeLoader.loadLibrary(RuntimeLoader.java:78)
        at edu.wpi.cscore.CameraServerCvJNI.forceLoad(CameraServerCvJNI.java:58)
        at edu.wpi.first.shuffleboard.plugin.cameraserver.CameraServerPlugin.onLoad(CameraServerPlugin.java:55)
        at edu.wpi.first.shuffleboard.app.plugin.PluginLoader.load(PluginLoader.java:303)
        at edu.wpi.first.shuffleboard.app.Shuffleboard.init(Shuffleboard.java:84)
        at com.sun.javafx.application.LauncherImpl.launchApplication1(LauncherImpl.java:824)
        at com.sun.javafx.application.LauncherImpl.lambda$launchApplication$2(LauncherImpl.java:195)
        at java.base/java.lang.Thread.run(Thread.java:834)
```

ThadHouse, 2019-10-26:
Closed by #609

closed 2019-10-26 by ThadHouse