wpilibsuite/shuffleboard #639: SendableChooser empty on startup instead of showing default value

trevnels, 2019-12-27:
**Describe the bug**  
When Shuffleboard starts after robot code is restarted, Sendable Choosers with a default value set programmatically do not show this default value on the dashboard. They instead show empty with a ! symbol. Internally, the chooser's active value is the default, but this is not represented on Shuffleboard.

ComboBox Choosers show completely blank with the !. SplitBox Choosers have the correct item highlighted but still show a !. When switching from Combo to Split and back to Combo, the correct item is indeed in the box, but still has the !.

**To Reproduce**  
Steps to reproduce the behavior:
1. Create a SendableChooser with a default option, and send it to Shuffleboard
2. With Shuffleboard closed, start robot code
3. Open Shuffleboard

The SendableChooser will be blank instead of showing the default/active value.

**Expected behavior**  
When Shuffleboard starts, SendableChooser widgets should be set to the currently active value instead of being empty.

**Screenshots**  
[Imgur Album](https://imgur.com/a/qUds1mf)

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Java version 11.0.1+13-LTS
 - Version 2020.1.1-beta4

**Additional Context**
My guess is that some sort of an update method isn't being called when the widget is created when it needs to be.

trevnels, 2019-12-27:
I believe the issue is that the listener [here](https://github.com/wpilibsuite/shuffleboard/blob/master/plugins/base/src/main/java/edu/wpi/first/shuffleboard/plugin/base/widget/ComboBoxChooserWidget.java#L43) is not being called on widget initialization. If I can fix it myself, I'll open a PR.

trevnels, 2019-12-27:
Fixed in #640

sciencewhiz, 2019-12-27:
Thanks for the PR. However, the fixed issues shouldn't be closed until the PR is merged.

PR #640 (2019-12-29): Make SendableChooser widgets default to the Chooser's default value if no selection has been made
# Overview
Fix #639 by making SendableChooser widget selection default to its default value instead of empty if a selection hasn't been made. If no default is specified, it will do the old behavior of showing an empty combo/split box.

# Screenshots
[Imgur Link](https://imgur.com/a/iEUBr09)
commit aef901d0 (2019-12-29): Make SendableChooser widgets default to the Chooser's default value if no selection has been made (#640)

If no default is specified, it will do the old behavior of showing an empty combo/split box.

Fixes #639
closed 2019-12-29 by PeterJohnson