wpilibsuite/shuffleboard #643: NullPointerException When Loading Themes from Plugin

Russell-Newton, 2020-01-24:
**Describe the bug**  
I am trying to create a plugin that includes a custom theme. When I place the .jar file into the plugins folder, Shuffleboard crashes when opening. The log file says that there is a `NullPointerException` that occurs when I create a new Theme object. I believe that it is related to the `localClass` parameter of the Theme constructor.

**To Reproduce**  
1. Place a plugin .jar file into ~/Shuffleboard/plugins
          * The plugin must attempt to create a new Theme object and export it with `getThemes()`.
2. Attempt to open Shuffleboard

**Expected behavior**  
I would expect that the plugin can load properly. Even if it can't find the style sheet, I wouldn't expect there to be a `NullPointerException` in the Theme constructor.

**Screenshots**  
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 10.0 amd64
 - Java version 11.0.4+11 -- AdoptOpenJDK
 - Version 2020.2.2

**Additional context**  
The same error occurs when I try to load in the example custom-theme plugin.

carbotaniuman, 2020-02-01:
Closed by #645. Disposition to close @PeterJohnson

carbotaniuman, 2020-08-24:
@PeterJohnson Disposition to close.

closed 2020-08-24 by PeterJohnson