wpilibsuite/shuffleboard #646: Smartdashboard.selectTab() continuously selects the tab

agasser, 2020-01-31:
**Describe the bug**  
After using `selectTab()` to select a tab with code, the user is not able to keep a different tab selected.

**To Reproduce**  
1. Use this code in a robot to create and select a tab:
```
  ShuffleboardTab driverTab = Shuffleboard.getTab("Driver");

  public void teleopInit() {
    Shuffleboard.selectTab("Driver");
  }
```
2. Deploy the code, then enable teleop mode.

The Driver tab does become selected, but when the user selects a different tab, the selected tab returns to Driver after a moment.

**Expected behavior**  
The chosen tab gets selected when `selectTab()` is called, but the user is still able to manually chose a different tab. If this is not the expected behavior, there is no reason to have other tabs because they are not usable.

**Desktop (please complete the following information):**
Version 2020.2.2 built on January 16, 2020
JRE: 11.0.4+11 -- AdoptOpenJDK
Operating system: Windows 10 10.0 amd64

Daltz333, 2020-02-01:
I mean, your continuously selecting the tab... Have a boolean toggle to confirm if set or not, or base it on NT

agasser, 2020-02-01:
Please explain how calling `selectTab()` one time in `teleopInit()` is continuously selecting the tab. Is something missing in the documentation that I have to call *after* `selectTab()` to stop continuously selecting it.

What would a boolean toggle be used for, or what action would I take base on the NT? It's only called once, so why would I need logic?

The intended behavior with the example code is that when teleop is activated, the *Driver* tab becomes selected. This should not prevent the user from selecting a different tab after `teleopInit()`.

Daltz333, 2020-02-01:
My apologies, I must be tired 😟
I read it at teleopPeriodic somehow. This is indeed an issue.

agasser, 2020-02-01:
No worries! Thanks for looking into it.

carbotaniuman, 2020-02-03:
selectTab() basically sets a NetworkTables entry to the title of the tab. Shuffleboard uses this to select a tab and force it to always be this tab. I agree updating documentation would probably be a smart move, maybe call it `forceTab()` and have an `unforceTab()` method?

agasser, 2020-02-03:
Would it be possible for Shuffleboard to also set this NT entry when the user selects a tab? This seems like it would solve the problem.
With this approach, you could also tell what tab is currently selected from the client end.

carbotaniuman, 2020-02-04:
Would there be any unintended effects from that? I don't know enough about this codebase to comment on that,

carbotaniuman, 2020-07-29:
I just tested this and it works fine.

carbotaniuman, 2020-08-24:
@PeterJohnson Disposition to close.

closed 2020-08-24 by PeterJohnson