wpilibsuite/shuffleboard #648: Update wiki to include importing JavaFX instructions

28andrew, 2020-02-02:
**Is your feature request related to a problem? Please describe.**  
Last year when I was making a ShuffleBoard plugin, FRC still used JDK 8 which included JavaFX. However, with the update to Java 11, JavaFX is no longer included. It's confusing now to figure out how to include JavaFX when making a ShuffleBoard plugin. WPILib projects that use JavaFX have proprietary plugins and an Artifactory repository hosting JavaFX. Below is a `build.gradle` that is an example of how JavaFX can be included without WPI gradle plugins. I'm not sure of how to use the WPI gradle plugins to import JavaFX.

```
plugins {
    id "com.google.osdetector" version "1.6.2"
}

ext.platform = osdetector.os == 'osx' ? 'mac' : osdetector.os == 'windows' ? 'win' : osdetector.os

apply plugin: 'java'

repositories {
    mavenCentral()
    jcenter()
    maven { url "https://jitpack.io" }
    maven { url 'https://frcmaven.wpi.edu/artifactory/release/' }
}

dependencies {
    compile group: "edu.wpi.first.shuffleboard", name: "api", version: '2020.2.2'
    compile "org.openjfx:javafx-base:11.0.1:$platform"
    compile "org.openjfx:javafx-graphics:11.0.1:$platform"
    compile "org.openjfx:javafx-controls:11.0.1:$platform"
    compile "org.openjfx:javafx-fxml:11.0.1:$platform"
}

task installPlugin(type: Copy) {
    from(tasks.named("jar"))
    into("${System.getProperty("user.home")}/Shuffleboard/plugins")
    description = "Builds the plugin JAR and installs it in the Shuffleboard plugins directory."
    group = "Shuffleboard Plugin"
}
```

**Describe the solution you'd like**  
Update wiki with method to include JavaFX, whether using WPI's proprietary gradle plugins or not (see: javafx method from a WPI gradle plugin used in kotlin build.gradle's of OutlineViewer and ShuffleBoard).

carbotaniuman, 2020-02-03:
You do not need JavaFX binaries for Shuffleboard plugins.

```kotlin
plugins {
    kotlin("jvm") version "1.3.31"
    id("org.openjfx.javafxplugin") version "0.0.8"
}

javafx {
    modules("javafx.controls", "javafx.fxml", "javafx.controls", "javafx.swing")
    configuration = "compileOnly"
}
```
These two blocks are all you really need. You just need the APIs to compile against, not the executables themselves as Shuffleboard provides them at runtime.

carbotaniuman, 2020-08-25:
Documented [here](https://docs.wpilib.org/en/latest/docs/software/wpilib-tools/shuffleboard/custom-widgets/creating-plugins.html).

Disposition to close @PeterJohnson

closed 2020-08-26 by AustinShalit