wpilibsuite/shuffleboard #651: Shuffleboard crashing upon loading custom plugin

kooryan, 2020-02-13:
**Describe the bug**  
When loading a custom plugin and opening Shuffleboard, the application crashes. 

**To Reproduce**  
Steps to reproduce the behavior:
1. Drag our custom into /Shuffleboard/plugins
2. Open Shuffleboard with the custom plugin

**Expected behavior**  
Shuffleboard should be able to load the plugin fine without any issues – this was the case last year. This year we updated the dependencies and requirements for the new version of Shuffleboard (2020.2.2) but upon loading the plugin the application still crashes.

Here is the error:
[shuffleboard.2020-02-12-20.41.22.log](https://github.com/wpilibsuite/shuffleboard/files/4196264/shuffleboard.2020-02-12-20.41.22.log)

Plugin class : 

``` 
@Description(group = "edu.wpi.first.shuffleboard", name = "Green Machine Plugin", version = "1.1.0", summary = "Base Plugin for all Green Machine Widgets")
public class Plugin1816 extends Plugin {

    public Plugin1816() {
        super();
        CoordinateWidget.setRobot(true);
    }

    @Override
    public List<ComponentType> getComponents() {
        return ImmutableList.of(
                WidgetType.forAnnotatedWidget(CoordinateWidget.class),
                WidgetType.forAnnotatedWidget(GstreamerWidget.class));
    }
}  
```

build.gradle:

```
repositories {
    maven {
        url 'https://plugins.gradle.org/m2/'
    }
    maven { url "https://jitpack.io" }
    maven { url "https://frcmaven.wpi.edu/artifactory/release/"}
    mavenCentral()
}

dependencies {
    compile group: "edu.wpi.first.shuffleboard", name: "api", version: "2020.2.2"
    compile group: 'edu.wpi.first.shuffleboard.plugin', name: 'networktables', version: getShuffleboardVersion()
    // https://mvnrepository.com/artifact/org.freedesktop.gstreamer/gst1-java-core
    // https://mvnrepository.com/artifact/org.freedesktop.gstreamer/gst1-java-fx
    compile group: 'org.freedesktop.gstreamer', name: 'gst1-java-fx', version: '0.9.0'
}

wrapper {
    gradleVersion = '6.1'
}

task installPlugin(type: Copy) {
    from(tasks.named("jar"))
    into("${System.getProperty("user.home")}/Shuffleboard/plugins")
    description = "Builds the plugin JAR and installs it in the Shuffleboard plugins directory."
    group = "Shuffleboard Plugin"
} 
```

Russell-Newton, 2020-02-13:
What's your code/project structure look like?

kooryan, 2020-02-13:
> What's your code/project structure look like?

https://github.com/TheGreenMachine/1816Plugin

Russell-Newton, 2020-02-13:
My understanding of the current proper plugin structure is that it should look something like the following:
```
src/main/
     java/some/identifier/plugin/
         data/
             MyDataType.java
             MyData.java
         widget/
             MyWidget.java
         MyPlugin.java
     resources/some/identifier/plugin/
         MyResource.css
```
Where the `@Description` on MyPlugin looks like
`@Description(group="some.identifier", name="My Plugin",  version="1.0.0", description="Something about My Plugin")`

You'll want to override the `getDefaultComponents()` and `getDataTypes()` methods too.

Another consideration is that you probably want you `settings.gradle` file to have a different root project name than JavaFX-11. That's just the name the jar gets when built.

Anything in the resources directory, when built, will get placed into the corresponding folder in the java directory. That is, when built, the jar will have `MyPlugin.java` and `MyResource.css` in the same folder, along with the data and widget subdirectories.

You probably don't actually need the data and widget subdirectories, but I'm fairly confident that everything needs to be in a plugin subdirectory and that the group in the  plugin description should reflect any directories that you place the plugin directory in.
I believe the reasoning for this is so that the reflection procedures can properly locate your Plugin class.

I know it's a lot of changes, but that will probably solve some of the problems with upgrading to 2020.

Russell-Newton, 2020-02-13:
I could be completely wrong. At the very least though, you shouldn't have the group name be "edu.wpi.first.shuffleboard".

kooryan, 2020-02-14:
So the Description error is still there, but also I am getting another saying that `module-info.class` is missing—still not sure how to fix either bug. 

[shuffleboard.2020-02-13-14.25.33.log](https://github.com/wpilibsuite/shuffleboard/files/4204823/shuffleboard.2020-02-13-14.25.33.log)

Russell-Newton, 2020-02-14:
You solved the problem where the PluginLoader would throw a, InvalidPluginException. 

I don't think that the remaining issue with the NoClassDefError is related to your plugin. I was able to successfully load your 1816Plugin-2.2.0.jar file that you have in your project's releases. I believe this may be an issue with your copy of shuffleboard. You should try completely uninstalling your current (and any previous) copy and reinstalling the latest one with the installer from <a href=https://docs.wpilib.org/en/latest/docs/getting-started/getting-started-frc-control-system/wpilib-setup.html>wpilib</a>, by selecting the Tools and Utilities box. Beyond that, I'm not sure how to fix the problem.

richiksc, 2020-02-14:
@Russell-Newton the 2.2.0 on our releases page is very outdated. We just tried that jar and it works, so our Shuffleboard installation is fine.

Russell-Newton, 2020-02-15:
Can you try editing your `build.gradle` file by commenting out your `jar` task and the `id 'application'` line in the `plugins` task and then rebuilding your jar? I've tried, but I haven't been able to load for a reason with AppSink requirements.

carbotaniuman, 2020-02-18:
Here's my attempt at it.

https://github.com/4639RoboSpartans/gstreamer_shuffleboard/blob/master/build.gradle.kts

Note the lack of `id("application")` as well as the JNA dependency.

Russell-Newton, 2020-02-18:
> Here's my attempt at it.
> 
> https://github.com/4639RoboSpartans/gstreamer_shuffleboard/blob/master/build.gradle.kts
> 
> Note the lack of `id("application")` as well as the JNA dependency.

What is the error log you get?

richiksc, 2020-02-18:
@carbotaniuman Thanks, we were able to fix it by updating to this, based off your `build.gradle`:

https://github.com/TheGreenMachine/1816Plugin/blob/master/build.gradle

Most notably, we removed the `id "application"` plugin, made JavaFX and WPILib dependencies `compileOnly`, and updated the `jar` task to build from the `runtimeClasspath` configuration and exclude `META-INF`. However, now we are getting a new exception:

```
SEVERE: Uncaught exception on thread `JavaFX Application Thread`.
java.lang.UnsatisfiedLinkError: Could not load library: gstreamer
	at org.freedesktop.gstreamer.lowlevel.GstNative.load(GstNative.java:56)
	at org.freedesktop.gstreamer.lowlevel.GstNative.load(GstNative.java:46)
	at org.freedesktop.gstreamer.lowlevel.GstElementFactoryAPI.<clinit>(GstElementFactoryAPI.java:36)
	at org.freedesktop.gstreamer.ElementFactory.makeRawElement(ElementFactory.java:304)
	at org.freedesktop.gstreamer.ElementFactory.make(ElementFactory.java:297)
	at com.edinarobotics.plugin.widget.GstreamerWidget.<init>(GstreamerWidget.java:48)
```

Russell-Newton, 2020-02-18:
@richiksc Your new issue is no longer a Shuffleboard issue.

Apparently many other people have had issues with gstreamer not loading. It requires java code (implemented through gradle) and native code, which is what's missing. I'm not sure how to fix this problem, but I hope this points you towards the right direction.

carbotaniuman, 2020-02-20:
@richiksc Contact me on the FRC discord for details. I'm carbotaniuman, you can prob just ping.

carbotaniuman, 2020-08-24:
@PeterJohnson Disposition to close.

closed 2020-08-24 by PeterJohnson