wpilibsuite/shuffleboard #652: Shuffleboard wasting resources

shaked113021, 2020-02-20:
Our team uses shuffleboard to display values to the drivers. In the last couple of days, shuffleboard suddenly started taking crazy amounts of memory and processing power. Almost 90% cpu power and 8 gb of RAM and it also started lagging like crazy (even though by looking in outline viewer we can clearly see that they are updating normally)

Daltz333, 2020-02-20:
Are you using graphs of any kind?

shaked113021, 2020-02-21:
We are using a combo box for path choosing,   for everything else we use text views which are for displaying and retrieving data. No graphs.

sciencewhiz, 2020-02-21:
What about cameras? Can you post your save file?

shaked113021, 2020-02-21:
We don't save files nor cameras, for now we solved the problem, we accidentally used enableActuatorWidgets in an inappropriate place. Shuffleboard is still showing very high memory usage, using JProfiler I have discovered that it creates a lot of weakreferences,the number of which is growing abnormally.

Daltz333, 2020-02-21:
Hm. I can't seem to reproduce, do you have any test case?

AE173, 2020-02-22:
> Hm. I can't seem to reproduce, do you have any test case?

Yes, we added you as a collaborator to our repo. OnyxTronix2020

shaked113021, 2020-02-22:
> > Hm. I can't seem to reproduce, do you have any test case?
> 
> Yes, we added you as a collaborator to our repo. OnyxTronix2020

Please use branch develop as test case

nimrod46, 2020-02-22:
Contrary to what @shaked113021 said, it did not solved our problem, the shuffleboard seams to freeze after we enable our robot. 
@Daltz333 We really need a quick fix for that as we are getting close to our first competition.

Daltz333, 2020-02-23:
I can't seem to find anything obvious, but I ran your code on my desktop using the [WPILib Sim](https://docs.wpilib.org/en/latest/docs/software/wpilib-tools/robot-simulation/index.html) and my CPU usage never rose over 5% and RAM over 550MB.

There have been issues with LiveWindow and performance in the past, could you try disabling any LiveWindow related functions and seeing the performance then?

Specs:
Ryzen 5 1500x
GTX 970
DDR4 2800MHz RAM
SATA SSD

nimrod46, 2020-02-23:
@Daltz333 I don't think it's possible to reproduce the problem while simming.

How can we disable the LiveWondow completely?

AE173, 2020-02-23:
> I can't seem to find anything obvious, but I ran your code on my desktop using the [WPILib Sim](https://docs.wpilib.org/en/latest/docs/software/wpilib-tools/robot-simulation/index.html) and my CPU [...]

We managed to fix the issue using `LiveWindow.disableAllTelemetry();`.
Apparently the Shuffleboard gets stuck after _moving_ the robot (pushing the robot in disable mode as well).
F.Y.I, We tried defining our motors without using WPI_ classes which did not help :/.

Thank you all for your help, we will keep this issue open...

Daltz333, 2020-02-23:
So LiveWindow was the cause? Thanks.

XzibitGG, 2020-03-21:
Shuffleboard memory leaks could also be happening due to multiple ObservableValue objects being created but not garbage collected everytime a listener is triggered. 

From: https://docs.oracle.com/javase/8/javafx/api/javafx/beans/value/ObservableValue.html#addListener-javafx.beans.value.ChangeListener-

> The ObservableValue stores a strong reference to the listener which will prevent the listener from being garbage collected and may result in a memory leak. It is recommended to either unregister a listener by calling removeListener after use or to use an instance of WeakChangeListener avoid this situation.