wpilibsuite/shuffleboard #696: 2021 artifacts for edu.wpi.first.shuffleboard.plugin:networktables were never published

calcmogul, 2021-04-13:
See https://frcmaven.wpi.edu/artifactory/release/edu/wpi/first/shuffleboard/plugin/networktables/

Note how https://frcmaven.wpi.edu/artifactory/release/edu/wpi/first/shuffleboard/api/ has 2021 artifacts.

Originally reported by https://github.com/wpilibsuite/frc-docs/issues/1325#issuecomment-819048736

sciencewhiz, 2021-04-14:
bisect says it was broken by 89a5b625983f6fa2f1d93b540280e095d735d4bf

sciencewhiz, 2021-04-16:
edu.wpi.first.shuffleboard.plugin:cameraserver were not published either

Starlight220, 2023-03-19:
Is this still relevant? 2021 was two years ago.

sciencewhiz, 2023-03-19:
They still aren't published

Starlight220, 2023-03-19:
Is there any reason to publish them, two years later?
I doubt it.

sciencewhiz, 2023-03-19:
There isn't a reason to go back and publish 2021 artifacts. There is reason (plugins) to fix publishing so that current API artifacts are published.

Starlight220, 2023-03-19:
Oh, you're saying that current artifacts aren't being published. I see.

If so, how does *anything* work?