wpilibsuite/shuffleboard #718: Shuffleboard does not open on WPILib 2022.1.1 Kickoff Release

ThraceAerospace, 2022-01-10:
**Describe the bug**  
Shuffleboard does not open, Progress bar halts at "Loading Camera Server Plugin" and disappears. Task Manager shows no task

**To Reproduce**  
Steps to reproduce the behavior:
1. Ran Shuffleboard from Desktop icon, from Start Tool menu in VS Code, from VBS Script in C:/Users/Public/WPILIB/2022/tools/Shuffleboard.vbs

All resulted in the same behavior

**Expected behavior**  
Shuffleboard open and function normally

**Screenshots**  
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10 LTSC  Build 17763.rs5_release.180914-1434
 - Java version [e.g. 1.8.0_171] - Unknown, Can not access shuffleboard menu. About Java reports "Version 8 RELEASE 311 (build 1.8.0_311-b11)"
 - Version: WPI Lib 2022.1.1 Kickoff Release

**Additional context**  
Add any other context about the problem here.
Just updated from 2021 where Shuffleboard was functioning

modelmat, 2022-01-10:
There exists two versions of the startup script, the one in [the installer](https://github.com/wpilibsuite/WPILibInstaller-Avalonia/blob/main/files/ScriptBase.py) and the one in [GradleRIO](https://github.com/wpilibsuite/GradleRIO/blob/main/src/main/resources/ScriptBase.py).

The one in `GradleRIO`, installed by running <kbd>WPILib: Install tools from GradleRIO</kbd> or `./gradlew InstallAllTools` seems to have incorrect logic. 
The one in the installer does not, and can be used by running `~/wpilib/2022/tools/ToolsUpdater.py` or appropriate script per-platform, which will override the `GradleRIO` versions.

calcmogul, 2022-01-10:
I was using the installer script, not the GradleRIO script. So that's definitely _an_ issue, but not this issue.

calcmogul, 2022-01-10:
If we can't figure it out, here's some opaque shell script that worked for me that starts the .jar as a background process and prints stdout and stderr to console. It assumes we'll only ever use the WPILib JDK.
```bash
#!/bin/bash
../jdk/bin/java -jar ${0/.sh/.jar} &
```
`$0` expands to the name of the script. `${0/.sh/.jar}` replaces instances of the string ".sh" in the variable `$0` with the string ".jar".

modelmat, 2022-01-11:
The scripts as-is in the repository both work at commit https://github.com/wpilibsuite/shuffleboard/commit/60c9c2a543a3cedaa3e250400f0bccac5ada0386 ([beta 1](https://frcmaven.wpi.edu/ui/native/development/edu/wpi/first/shuffleboard/shuffleboard/2022.1.1-beta-1)) but break at commit https://github.com/wpilibsuite/shuffleboard/commit/59567416174c78809f3fd4f3c83a1d0acd832f2b ([beta 1-1](https://frcmaven.wpi.edu/ui/native/development/edu/wpi/first/shuffleboard/shuffleboard/2022.1.1-beta-1)).

calcmogul, 2022-01-11:
The second one fails with an unsatisfied link error, which has since been fixed.

sciencewhiz, 2022-01-11:
@ThraceAerospace can you post the shuffleboard log file in `C:\Users\[User]\Shuffleboard`

@calcmogul @modelmat Any issues with python script would be unrelated to this issue which is with windows launched with the vbs. I recommend creating a new issue.

calcmogul, 2022-01-11:
I'm actually fairly confident at this point the issue is caused by changes to shuffleboard rather than something in the scripts invoking it. The 2021 shuffleboard jar and 2022 script works, while the 2022 shuffleboard jar and 2022 script doesn't.

ThraceAerospace, 2022-01-11:
@sciencewhiz

Here are the log files you asked for

[shuffleboard.2022-01-10-17.08.01.log](https://github.com/wpilibsuite/shuffleboard/files
[shuffleboard.202
[shuffleboard.2022-01-10-17.05.42.log](https://github.com/wpilibsuite/shuffleboard/files/7848948/shuffleboard.2022-01-10-17.05.42.log)
2-01-10-17.07.15.
[shuffleboard.2022-01-10-17.05.27.log](https://github.com/wpilibsuite/shuffleboard/files/7848949/shuffleboard.2022-01-10-17.05.27.log)
log
[shuffleboard
[shuffleboard.2022-01-10-17.03.15.log](https://github.com/wpilibsuite/shuffleboard/files/7848952/shuffleboard.2022-01-10-17.03.15.log)
.2022-01-10-17.03.26.log](https://github.com/wpilibsuite/shuffleboard/files/7848951/shuffleboard.2022-01-10-17.03.26.log)
](https://github.com/wpilibsuite/shuffleboard/files/7848946/shuffleboard.2022-01-10-17.07.15.log)
/7848945/shuffleboard.2022-01-10-17.08.01.log)
[shuffleboard.2022-01-10-17.06.42.log](https://github.com/wpilibsuite/shuffleboard/files/7848947/shuffleboard.2022-01-10-17.06.42.log)

sciencewhiz, 2022-01-12:
@ThraceAerospace does it work if you run from the command line `c:\Users\Public\wpilib\2022\jdk\bin\java -jar c:\Users\Public\wpilib\2022\tools\shuffleboard.jar`

ThraceAerospace, 2022-01-12:
@sciencewhiz That worked as exspected

Command Output:

Microsoft Windows [Version 10.0.17763.2452]
(c) 2018 Microsoft Corporation. All rights reserved.

C:\Users\Public\wpilib\2022\jdk\bin\java -jar c:\Users\Public\wpilib\2022\tools\ShuffleBoard.jar
Jan 12, 2022 6:45:12 PM edu.wpi.first.shuffleboard.app.Loggers setupLoggers
CONFIG: Configuration done.
Jan 12, 2022 6:45:12 PM edu.wpi.first.shuffleboard.app.Loggers setupLoggers
CONFIG: Shuffleboard app version: 2022.1.1
Jan 12, 2022 6:45:12 PM edu.wpi.first.shuffleboard.app.Loggers setupLoggers
CONFIG: Running from C:\Users\Public\wpilib\2022\tools\ShuffleBoard.jar
Jan 12, 2022 6:45:12 PM edu.wpi.first.shuffleboard.app.Shuffleboard init
INFO: Build time: 2022-01-07T04:34:12.039975700Z
Jan 12, 2022 6:45:12 PM edu.wpi.first.shuffleboard.app.plugin.PluginLoader load
INFO: Loading plugin edu.wpi.first.shuffleboard:Base:1.3.0
Jan 12, 2022 6:45:12 PM edu.wpi.first.shuffleboard.app.plugin.PluginLoader load
INFO: Loaded plugin edu.wpi.first.shuffleboard:Base in 89ms
Jan 12, 2022 6:45:12 PM edu.wpi.first.shuffleboard.app.plugin.PluginLoader load
INFO: Loading plugin edu.wpi.first.shuffleboard:NetworkTables:2.3.0
Jan 12, 2022 6:45:12 PM edu.wpi.first.shuffleboard.app.plugin.PluginLoader load
INFO: Loaded plugin edu.wpi.first.shuffleboard:NetworkTables in 15ms
Jan 12, 2022 6:45:12 PM edu.wpi.first.shuffleboard.app.plugin.PluginLoader load
INFO: Loading plugin edu.wpi.first.shuffleboard:CameraServer:3.1.0
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
Jan 12, 2022 6:45:16 PM edu.wpi.first.shuffleboard.app.plugin.PluginLoader load
INFO: Loaded plugin edu.wpi.first.shuffleboard:CameraServer in 3134ms
Jan 12, 2022 6:45:16 PM edu.wpi.first.shuffleboard.app.MainWindowController initialize
INFO: Setting up plugins in the UI
Jan 12, 2022 6:45:16 PM edu.wpi.first.shuffleboard.app.Shuffleboard init
INFO: Took 451ms to load the main FXML
NT: connect() to localhost port 1735 timed out
SLF4J: No SLF4J providers were found.
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#noProviders for further details.
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
Jan 12, 2022 6:45:19 PM edu.wpi.first.shuffleboard.app.Shuffleboard start
WARNING: Took 7139ms to start Shuffleboard
Jan 12, 2022 6:45:19 PM edu.wpi.first.shuffleboard.app.json.WidgetSaver deserialize
WARNING: Saved source type is not present, adding destroyed source(s) instead
Jan 12, 2022 6:45:19 PM edu.wpi.first.shuffleboard.app.json.WidgetSaver deserialize
WARNING: Saved source type is not present, adding destroyed source(s) instead
Jan 12, 2022 6:45:19 PM edu.wpi.first.shuffleboard.app.json.WidgetSaver deserialize
WARNING: Saved source type is not present, adding destroyed source(s) instead
Jan 12, 2022 6:45:19 PM edu.wpi.first.shuffleboard.app.json.WidgetSaver deserialize
WARNING: Saved source type is not present, adding destroyed source(s) instead
Jan 12, 2022 6:45:19 PM edu.wpi.first.shuffleboard.app.json.WidgetSaver deserialize
WARNING: Saved source type is not present, adding destroyed source(s) instead
Jan 12, 2022 6:45:19 PM edu.wpi.first.shuffleboard.app.json.WidgetSaver deserialize
WARNING: Saved source type is not present, adding destroyed source(s) instead
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out
NT: connect() to localhost port 1735 timed out

modelmat, 2022-01-13:
By 

> That worked as exspected

Do you mean, it works, or it works the same way the VBS script did?

If the former, then the issue is what is mentioned [here](https://github.com/wpilibsuite/shuffleboard/issues/718#issuecomment-1009481832) as there was a change in how return codes seem to be handled, which affects VBS and python scripts.

If it still "closes" after reaching that stage (the same way it did with the VBS script) then there is a different issue to what we think it is, and the log isn't helpful because there is no mention of a crash.

ThadHouse, 2022-01-13:
We found the root cause of the issue here. it's an issue in the vbs files, but caused by a change in wpiutil. We're working on what the right fix is.

For more explanation, the script when it closes closes the stdout it created. Our old print impl in wpiutil just ignored a broken pipe. The new one throws a c++ exception, which causes the program to be aborted. When you launch Java directly, the pipe stays alive so any prints don't crash.

ThraceAerospace, 2022-01-13:
@modelmat Sorry for the bad choice of words

Running the java jar command opened Shuffleboard and after that it worked normally and with no bugs

sciencewhiz, 2022-01-16:
Can you try replacing `c:\Users\Public\wpilib\2022\tools\shuffleboard.jar` with the file linked below and then run with the shortcut? This has the candidate change in wpiutil that was mentioned above.

https://wpilib.jfrog.io/artifactory/wpilib-generic-local/shuffleboard-test/shuffleboard-allwpilib3919/shuffleboard.jar

sciencewhiz, 2022-01-16:
This fix was included in WPILib 2022.2.1 release

ThraceAerospace, 2022-01-16:
@sciencewhiz Alright ty, already updated to 2022.2.1

PeterJohnson, 2022-01-17:
Closing, please reopen if you still have an issue.

closed 2022-01-17 by PeterJohnson