wpilibsuite/shuffleboard #719: shuffleboard won't start

jpfrigeriogithub, 2022-01-12:
Hi!  I have followed the directions for installing the WPILib tools here:
https://docs.wpilib.org/en/stable/docs/zero-to-robot/step-2/wpilib-setup.html

But shuffleboard won't start, crashes every time.  It flashes some messages ("loading camera server", etc), but then quits.

This is a fully updated Windows 10 laptop.
I have fully uninstalled all previous versions.
I have updated java and .net, all are current.
I have uninstalled and re-installed a few times with several reboots.
Nothing seems to help.
I have also checked the windows event viewer to see if the exception thrown was caught there, but it's blank.

Glad for any tips you might have.  
Thanks.

ThadHouse, 2022-01-12:
Just to double check, you're not using an N version of windows, right?

jpfrigeriogithub, 2022-01-12:
No, straight windows 10, came preinstalled.  64-bit.

sciencewhiz, 2022-01-12:
Does it work if you run from the command line `c:\Users\Public\wpilib\2022\jdk\bin\java -jar c:\Users\Public\wpilib\2022\tools\shuffleboard.jar`

jpfrigeriogithub, 2022-01-12:
Yes, thanks!

I guess that means the problem is in the shuffleboard.vbs file that the
shortcut calls.

I tried to run that from the command line to see if it gave any helpful
diagnostics.  But it didn't, it failed just as before and spit out nothing
to the console at all.

Hopefully I'm a one-off and it's not a bug affecting too many people. :-)

thanks again.

calcmogul, 2022-01-12:
We've narrowed this down to the application crashing after the script exits when the application can't write to stdout or stderr. The Python script can be fixed by setting stdout and stderr to subprocess.DEVNULL, but I don't know what the equivalent for VBS is.

sciencewhiz, 2022-01-16:
Can you try replacing `c:\Users\Public\wpilib\2022\tools\shuffleboard.jar` with the file linked below and then run with the shortcut? This has a candidate change to not write to stdout/stderr as mentioned above.

https://wpilib.jfrog.io/artifactory/wpilib-generic-local/shuffleboard-test/shuffleboard-allwpilib3919/shuffleboard.jar

sciencewhiz, 2022-01-16:
This fix was included in WPILib 2022.2.1 release

PeterJohnson, 2022-01-17:
Closing, please reopen if there is still an issue.

closed 2022-01-17 by PeterJohnson