wpilibsuite/shuffleboard #720: Graphs get duplicate entries on code deploy

randomstring, 2022-01-13:
**Describe the bug**  
When graphing network table values after redeploying code, the graph will have two copies of the same NT value with the same name. I first discovered this when graphing two values on one graph, but the bug is present even when values are in separate graphs. 

In my code I publish two doubles with this code.

https://github.com/FRC-Sonic-Squirrels/swerve-template/blob/0c9cf83f6509f439c864c6501059f33c5e1afcee/src/main/java/frc/robot/commands/DriveWithSetRotationCommand.java#L119-L120

**To Reproduce**  
Steps to reproduce the behavior:
1. publish value to shufflboard (see link to example code above)
2. create entry on Shuffboard page and choose "disaplay as graph"
3. observe graph works
4. redeploy code from VS Code
5. now there are two numbers graphed with identical NT names
6. redeploy code
7. now the graph has 3 identical NT entries all graphed
8. repeat

**Expected behavior**  
graph should only have one copy of the Network Table value in the graph

**Screenshots**  
See screen shot
[image]

**Desktop (please complete the following information):**
 - OS: Windows
  - Roborio 1.0, Image FRC_roborio_2022_v3.0
  - DS 22.0
  - Lib Java 2022.1.1
  - Pidgeon 22.0
  - PDP 1.40
  - CANCoder 22.0.0.0
  - TalonFX 22.0.0.0

**Additional context**  
It feels like Shuffleboard is using more CPU and memory than prior to the 2022.1.1 release, I don't have a stats to back this up. Just the impression that on the same laptop with a far more complex Shuffleboard layout on the 2021 code base, Shuffleboard didn't cause my laptop to start "weezing" (i.e. run the fan nonstop and kill the battery).

PeterJohnson, 2022-01-17:
Compared to which previous version--2021, 2020, 2019?  What widgets are you using?  What does the task manager show for CPU/GPU usage?

peter-popescu, 2022-02-11:
I've had the same issue since last year, but I always thought it was a problem with my implementation of SmartDashboard.putNumber. Glad I've found someone else with this issue, not sure how I haven't seen a CD post about it. W.r.t. the performance issues, I've found that ShuffleBoard starts maxing my CPU usage after I've redeployed my code a lot and SB starts struggling to plot 15 lines of the same data simultaneously...

bovlb, 2022-12-07:
I believe this problem may stem from [the implementation of `getNumberSeries`](https://github.com/wpilibsuite/shuffleboard/blob/e08ac2754826abd464aa89f14b61a5264f47e342/plugins/base/src/main/java/edu/wpi/first/shuffleboard/plugin/base/widget/GraphWidget.java#L326).  We're using the source object as the key in the map, and apparently that is changing identity when the code is redeployed.  We should never be adding a new series when receiving an update for an existing series.

Suggested mitigations:
* Make numberSeriesMap map from the series name (networktable key) instead of from the source object.
* Add an `equals()` method (and also `hashCode()`) to [`DataSource`](https://github.com/wpilibsuite/shuffleboard/blob/main/api/src/main/java/edu/wpi/first/shuffleboard/api/sources/DataSource.java) that uses `getName()`.

Similarly in `getArraySeries`.

PR #774 (2023-03-27): Add DataSource.equals()
Should fix #720, and maybe even the disconnected source issues.
closed 2023-03-27 by PeterJohnson