wpilibsuite/shuffleboard #724: Field2d widget does not autopopulate

agasser, 2022-03-14:
**Describe the bug**  
Adding the Field2d widget to shuffleboard does not automatically populate it in the view. The user always has to find it in the NetworlTables and drag into the view.

**To Reproduce**  
Steps to reproduce the behavior:
1. Create a robot project with this code:
```
    var field = new Field2d();
    SmartDashboard.putData(field);
```
2. Run the project and connect with Shuffleboard

**Expected behavior**  
The Field2d widget should appear on the SmartDashboard tab automatically.

The Field2d widget is available in the NetworkTables, but the user has to drag the entry to the view for it to appear.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Java version 11.0.13+8
 - Version 2022.4.1

**Additional context**  
When the widget is added to a tab using the Shuffleboard API is also does not auto-populate. It has to be dragged by the user.
```
    var field = new Field2d();
    var driverTab = Shuffleboard.getTab("Driver");
    driverTab.add("Field", field);
```

PeterJohnson, 2022-03-14:
@noamzaks

noamzaks, 2022-03-17:
I think it needs to be registered as the default widget for the field type in https://github.com/wpilibsuite/shuffleboard/blob/main/plugins/base/src/main/java/edu/wpi/first/shuffleboard/plugin/base/BasePlugin.java like the other widgets.

I'll verify this works later today

PR #725 (2022-03-20): fix: population of field widget
Fixes #724.

Tested on Ubuntu in simulation
closed 2022-03-20 by PeterJohnson