wpilibsuite/shuffleboard #726: Shuffleboard appears to have memory leak

jee7s, 2022-03-20:
Possibly related to #721 

**Describe the bug**  
Running shuffleboard for a long time (20 mins) results in apparent memory leak. Initial memory utilization starts at about 400MB, growing to 2400MB. 

**To Reproduce**  
Start Shuffleboard with robot disabled with disabled robot code reporting values. We report status of DIOs and some motors. We notably do not have cameras connected to the RIO when Shuffleboard includes camera displays. After consuming sufficient RAM, PC CPU utilization rises, resulting in long trip times >70ms and occasional intermittent packet loss. The packet loss is sufficient to briefly disable the robot for about 200ms

**Expected behavior**  
Shuffleboard should not leak memory.

**Screenshots**  
Will provide at next opportunity 

**Desktop (please complete the following information):**
 - OS: Windows
 - Java version: OpenJDK version to be reported 
 - Shuffleboard version: 2022.4.1

**Additional context**  
This problem is difficult to detect at the field. FMS reports the long trip time and CPU utilization can be observed. But the root cause is not apparent without opening task manager.

PeterJohnson, 2022-03-20:
In order to help us try to reproduce this, please upload your shuffleboard.json file and point us to your robot code.  Going to Help | About in Shuffleboard will confirm the Java version (JRE) and Shuffleboard version.

Daltz333, 2022-03-21:
Additionally, can you share task manager screenshots when the memory leak has occurred?

jee7s, 2022-03-21:
Robot code is here: https://gitihub.com/frc6357/robot_code_2022 on branch 'master'

JRE is 11.0.13+8 Eclipse Adoptium

[DriverStation2022.json.txt](https://github.com/wpilibsuite/shuffleboard/files/8318300/DriverStation2022.json.txt)

Extension change on JSON file to comply with github.com extension policy.

jee7s, 2022-03-21:
Attached are screen shots of Task Manager
[image]
[image]

During this test we did notice a correlation between having a graph visible on the Shuffleboard and the memory growth. We've read elsewhere that graphs consume more memory. 

Is it possible we are observing an expected increase in memory usage due to the graph implementation? Does the graph hold on to more than the visible 30 seconds of data? If this is the expected behavior, then this bug may boil down to documentation for the possibility the laptop resources will be consumed and could result in the packet loss and intermittent disable behavior.

ThadHouse, 2022-03-21:
So something interesting about Java is it won't garbage collect until either the system says there is memory pressure, or a percentage of your memory (about 1/4) is used. So if you run on a system with a lot of memory, it will just keep using memory, and won't collect. There are some java profiling tools you can use to check for these cases. Based on your task manager logs, I'd suspect thats what is happening, as it looks like you have a lot of memory in your system.