wpilibsuite/shuffleboard #735: Shuffleboard.startRecording() does not work when connected to FMS

agasser, 2022-09-27:
When connected to the FMS at a competition the  `Shuffleboard.startRecording()` method does not start recording. The method works great when connected directly to the robot in the shop but does not work at competitions.

**To Reproduce**  
Steps to reproduce the behavior:
1. Add `Shuffleboard.startRecording()` to `autonomousInit()`
2. Add `Shuffleboard.stopRecording()` to `disabledInit()`
3. Connect driverstation running Shuffleboard to the FMS
4. Connect robot to the FMS
5. Run an actual match

**Expected behavior**  
Recording data is created from Shuffleboard.

**Actual behavior**
No recordings are created.

**Desktop (please complete the following information):**
 - OS: Windows
 - Java version: 11 (bundled with WPILib)
 - Version: 2022.4.1

**Workaround**
The Drive Team can manually start the recording by pressing the record button before the match starts.

PeterJohnson, 2022-09-27:
To be clear, you're only expecting to record autonomous?  The stop in disabled will stop the recording after autonomous (as the robot will spend a short time in disabled mode before it goes enabled for teleop mode).

agasser, 2022-09-27:
@PeterJohnson That's good point, we would probably want to find a better stopping point. Even so, the bug still holds since no recording is created.

Starlight220, 2023-03-19:
I've tried reproducing this locally in sim (with the FMS checkbox enabled).
The NT topic changes correctly, and the start/stop recording buttons in the shuffleboard UI do change correctly. Recording files are created with data, though I don't know how to check if they're correct.
However, the shuffleboard-side console/log is spammed with these errors:
```
NT: ERROR: reached maximum number of multi-subscribers, not adding listener (LocalStorage.cpp:1108)
NT: ERROR: reached maximum number of listeners to '/Shuffleboard/.recording/RecordData', not adding listener (LocalStorage.cpp:1000)
```