wpilibsuite/shuffleboard #736: Shuffleboard crashes frequently on MacOS

bovlb, 2022-10-13:
**Describe the bug**  
Shuffleboard will start up, connect to the robot, and display results, but will then crash within a few minutes.  

**To Reproduce**  
Steps to reproduce the behavior:
1. Install latest WPILIB on MacOS
2. Launch shuffleboard from VS Code using "Start Tool"
3. Wait a few minutes

**Expected behavior**  
Shuffleboard should be usable for a three hour programming session.

**Desktop (please complete the following information):**
 - OS: Mac OS X 12.5.1 x86_64
 - Java version 11.0.13+8 Eclipse Adoptium
 - Version 2022.4.1

bovlb, 2022-10-13:
[Archive.zip](https://github.com/wpilibsuite/shuffleboard/files/9779629/Archive.zip)
diag files

bovlb, 2023-01-16:
I installed 2023.2.1 today and was able to run ShuffleBoard for a three-hour session without a single crash.  I didn't see anything in the change notes about crashing less, but this is an order of magnitude better.

Starlight220, 2023-03-19:
There were multiple changes between 2022 and 2023 regarding how macos artifacts are handled; I expect that's what fixed it.

Can this be closed?

bovlb, 2023-03-19:
I have not seen this issue in the 2023 season.

closed 2023-03-19 by bovlb