wpilibsuite/shuffleboard #745: Shuffleboard continuously jumps back to start tab when using selectTab and kGyro

tomjwinter, 2023-01-06:
**Describe the bug**  
When using WPILIB code to create the Shuffleboard layout and select the initial tab, if one of the tabs contains a BuiltinWidget.kGyro, then you will no be able to change tabs, because the tab will switch back to the start tab right away.
It does not seem to matter which tab the Gyro is on (start tab or not).

**To Reproduce**  
I created a very simple Robot.java to duplicate this problem.
Compile and deploy the attached Robot.java (you will also need the https://dev.studica.com/releases/2023/NavX.json vendordeps).

You may or may not need to restart shuffleboard after deploying (depending on your previous layout).
It will automatically start on Tab Two as expected.  But if you try to select Tab one, it will bounce back to Tab Two automatically.

Removing the kGyro widget eliminates the problem, so there is some sort of interaction with selectTab and kGyro.

**Desktop (please complete the following information):**
 - OS: Windows 10 10.0 amd64
 - Java version: 17.0.5+8 -- Eclipse Adoptium
 - Version: 2023.1.1 built on 1/2/23

**Additional context**  
[Robot_java.txt](https://github.com/wpilibsuite/shuffleboard/files/10361374/Robot_java.txt)

sciencewhiz, 2023-01-07:
I'm not able to duplicate this running in simulation

korozya-bit, 2023-03-12:
I've also ran into this exact problem..

Starlight220, 2023-03-12:
I can't duplicate this in simulation either. Very weird.

Starlight220, 2023-03-12:
... And, I've been able to reproduce this. Somehow, the gyro value change re-does the tab selection.

Starlight220, 2023-03-12:
Okay, this is very broken. It seems that *any* data change will reset the tab selection to the NT-selected one.

PR #771 (2023-03-27): Don't switch to selected tab on value change
Fixes #745

For some reason, the tab structure listeners were called also on value changes (and not just metadata changes). I've done some testing to ensure this doesn't break functionality, but I may have missed something.
closed 2023-03-27 by PeterJohnson