wpilibsuite/shuffleboard #752: MacOS Client Hangs When Changing NetworkTable address

randomstring, 2023-01-27:
I'm using the 2023.2.1 WPILib release on MacOS, Intel based. 

**Describe the bug**  
MacOS Client Hangs When Changing NetworkTable address. 

**To Reproduce**  
Steps to reproduce the behavior:
1. Open `Preferences`
2. Select `Plugins > NetworkTables`
3. change the `Server`

Specifically my default server is set to our team's robot IP: 10.29.30.2, clicking on it and then deleting the IP address character by character to replace it with "localhost" for testing with the robot sim. Before I can finish deleting the old IP and replace it, the GUI hangs with with a spinning "pizza wheel."

I'm guessing that Shuffleboard is trying to connect to an invalid server name ("10.29.30" in this case) and hanging. 

I did not have this problem on Windows 11. 

I found a workaround. If you copy the desired server name into your paste buffer, then select the old server name and click paste, you can change the server name in one action. When changed in this way Shuffleboard doesn't hang and it works as expected. 

**Expected behavior**  
Not to hang.

**Screenshots**  
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS x 13 x86_64
 - Java version: JRE 17.0.5+8
 - WPILib 2023.2.1

**Additional context**  
Add any other context about the problem here.

PeterJohnson, 2023-02-17:
This may be fixed by wpilibsuite/allwpilib#5102.

PeterJohnson, 2023-02-20:
Can you confirm if this is fixed with 2023.4.1?

randomstring, 2023-02-20:
Confirmed. Shuffleboard does not hang when changing the NT address.

Fixed.

closed 2023-02-20 by PeterJohnson