wpilibsuite/shuffleboard #768: Command activation and deactivation buttons do not work properly.

Tom-ne, 2023-02-21:
Hello, we have multiple commands in Shuffleboard.
We have encountered an isssue that the button the activates and deactivates the command, almost always does not work on the first try, almost every time we need to press the button more than one time.

**To Reproduce**  
Steps to reproduce the behavior:
1. add commands to shuffleboard
2. try to run them

**Desktop (please complete the following information):**
Version 2023.3.2
JRE: 17.0.5+8 -- Eclipse Adoptium
Operating system: Windows 10 10.0 amd64

hollisms, 2023-02-25:
Same issue on Mac M1

Starlight220, 2023-03-19:
Can you upload Shuffleboard's logs from times that this happened? Shuffleboard log files are at `~/Shuffleboard/`.

Tom-ne, 2023-03-19:
I think it might be this, but I am not sure
[shuffleboard.2023-02-13-19.05.06.log](https://github.com/wpilibsuite/shuffleboard/files/11011182/shuffleboard.2023-02-13-19.05.06.log)

Starlight220, 2023-03-19:
From a quick glance there seem to be multiple exceptions in the log. Will investigate more in a bit.

Starlight220, 2023-03-19:
@Tom-ne the log you uploaded is from Shuffleboard v2023.1.1.
The common exception there was fixed in #759 (released in 2023.4.1).
There's some more errors/warnings there (some problems with loading saved layouts and CameraServer properties), but doesn't seem to show anything obviously part of this issue.

Tom-ne, 2023-03-19:
I will check the log file on our computers again and try to find the log file

Tom-ne, 2023-03-20:
[shuffleboard.2023-02-21-10.19.13.log](https://github.com/wpilibsuite/shuffleboard/files/11020190/shuffleboard.2023-02-21-10.19.13.log)
I think it might this, near the end of the file

Tom-ne, 2023-03-21:
I dont know

Starlight220, 2023-03-29:
Does this reproduce with the newly-releases 2023.4.3?
We hope that https://github.com/wpilibsuite/shuffleboard/pull/778 fixed this.

Tom-ne, 2023-04-02:
I don't know, I need to check

chauser, 2023-07-24:
This should have been fixed by the fix for allwpilib#5323 but it should be checked. Will need to use allwpilib HEAD for the test. I am not able to do the test now but if someone can then this can be closed.