wpilibsuite/shuffleboard #770: Shuffleboard data disappears from widgets

Crossle86, 2023-03-02:
We have been seeing a problem with Shuffleboard that is still happening on 2023.4.2 (robot and SB). What we see is that after a period of time of successful deploys and test runs, you will approach the DS to run a test and SB widgets will be blank, though they had data moments before. Normally, the widgets always display the last data sent. We have to restart SB to get our widgets to display data. This happens one or two times a day. Haven't observed the data disappear, it's just gone the next time you look at it. Have seen this after DS PC has been closed overnight, but not always. Also observe this after a deploy, but no problem on most deploys. When the widgets are blank, SB indicates Network tables are connected. Been trying to define this issue further without success.

This is a scary proposition for competition.

Starlight220, 2023-03-19:
Can you upload Shuffleboard log files from when this happened? Shuffleboard log files are located at `~/Shuffleboard/`.

Crossle86, 2023-03-21:
I am watching for it do it again, will get the log...hasn't happened in a match but lots of times in the pits and our lab.

Starlight220, 2023-03-21:
In multiple logs I've seen there are messages that might indicate problems connecting between live NT sources and the sources in a layout file, but I'm not sure if that's this issue.

nihonjinrxs, 2023-03-27:
We've also seen this behavior, and have had a similarly hard time pinning it down. 

Might it be possible to put a button on the interface to hard re-sync connections to Network Tables if this happens? If that could be reliable, it could at least be a workaround for competition until we can figure out why it's happening.

If you think this is a good idea, can you point me to where in the code base to look? I'm happy to try to PR this, but it would be my first time working on this repo, so I'd need a bit of guidance to get started, particularly around code base organization and how best to test such a thing.

Starlight220, 2023-03-27:
Yep, there indeed is an informative error there. Something is throwing a `ConcurrentModificationException` when trying to reconnect a source.

Crossle86, 2023-03-28:
All sources do not display. The source you mention set very early in our start up. It and a number of other sources are created dynamically in code by our swerve code. The rest of the sources we display are in the SB layout file. Note we are using the NT3 support in the ShuffleBoard class.

closed 2023-03-28 by PeterJohnson

EddieDL, 2023-03-31:
I believe we have the latest version and are experiencing this issue still. Where can I find the Shuffleboard logs exactly and do I just need to get them when it happens?

Starlight220, 2023-03-31:
Shuffleboard's logs are located at `~/Shuffleboard/`. On Windows this would be `C:/Users/<yourusername>/Shuffleboard/`.
When the issue happens, close shuffleboard and upload the most recent log to here.

EddieDL, 2023-03-31:
It just happened to me and this is the most recent log file.
[shuffleboard.2023-03-31-12.20.36.log](https://github.com/wpilibsuite/shuffleboard/files/11124010/shuffleboard.2023-03-31-12.20.36.log)

Let me know if you need anything more.

Crossle86, 2023-04-04:
I'm just back from vacation so I have not observed a failure myself but my students updated the SB to the latest version and told me that they have observed the same failure. I don't have a log file at this time but will catch one the next time it fails.

PR #779 (2023-03-28): Use ConcurrentHashMap for static collection of NT sources
Might fix #770?
Seems like a very small/simple fix, but on the other hand this may be what's happening: the relevant exception in the log [here](https://github.com/wpilibsuite/shuffleboard/issues/770#issuecomment-1485491710) is a ConcurrentModificationException on this collection.

Not sure why this would break all sources, though. Maybe the CME is interrupting an iteration?