wpilibsuite/shuffleboard #797: Field2d widget causes exception every frame

feoranis26, 2024-03-15:
**Describe the bug**  
Log file gets spammed with error: 
```
Mar XX, 2024 XX:XX:XX XM edu.wpi.first.shuffleboard.plugin.networktables.sources.CompositeNetworkTableSource lambda$new$0
WARNING: Incomplete data for type Field2d
edu.wpi.first.shuffleboard.api.data.IncompleteDataException: Incomplete data in map: {.name=Odom, .type=Field2d, .controllable=true, Robot=[D@35f5daf9,  Localized=x: 1.4519350881997504, y: -31.02742566505306}
	at edu.wpi.first.shuffleboard.api.data.ComplexDataType.fromMap(ComplexDataType.java:33)
	at edu.wpi.first.shuffleboard.plugin.networktables.sources.CompositeNetworkTableSource.lambda$new$0(CompositeNetworkTableSource.java:58)
	at edu.wpi.first.shuffleboard.plugin.networktables.sources.NetworkTableSource.lambda$setTableListener$0(NetworkTableSource.java:81)
	at edu.wpi.first.shuffleboard.api.util.FxUtils.lambda$runOnFxThread$0(FxUtils.java:64)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$10(PlatformImpl.java:457)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:456)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:96)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$runLoop$3(WinApplication.java:184)
	at java.base/java.lang.Thread.run(Thread.java:833)
Caused by: java.lang.ClassCastException: class java.lang.String cannot be cast to class [D (java.lang.String and [D are in module java.base of loader 'bootstrap')
	at edu.wpi.first.shuffleboard.plugin.base.data.FieldData.<init>(FieldData.java:73)
	at edu.wpi.first.shuffleboard.api.data.ComplexDataType.fromMap(ComplexDataType.java:31)
	... 10 more
```

**To Reproduce**  
1. Send ANY Field2d widget using SmartDashboard from robot code
2. Exception gets thrown every frame

**Expected behavior**  
No exception gets thrown, and robot pose is correctly displayed

**Actual behavior**
Field2d widget causes exception (every frame?), and robot pose is fixed at 0, 0

**Screenshots**  
Also causes MASSIVE log file spam
10 GB+ log files:
[image]
Smaller log file:
[shuffleboard.2024-03-07-11.44.02.log](https://github.com/wpilibsuite/shuffleboard/files/14614965/shuffleboard.2024-03-07-11.44.02.log)

**Desktop:**
 - OS: Windows
 - Java version: 17.0.6
 - Version: 2024.2.1
[image]

Starlight220, 2024-03-15:
My guess is that the widget code doing the parsing doesn't match the Sendable implementation.
Can you send how the NT data looks (from any raw NT view)?

PeterJohnson, 2024-03-15:
There were several updates to the NT code in 2024.3.1.  Can you confirm it still happens on that version or later (2024.3.2 was just released)?

Starlight220, 2024-03-15:
What is the type of "Localized"?
It should be a `double[3]`.

Starlight220, 2024-03-15:
Can you please test a build from #798 and see that the issue doesn't reproduce and no other issues pop up?

SamCarlberg, 2024-03-15:
> Incomplete data in map: {.name=Odom, .type=Field2d, .controllable=true, Robot=[D@35f5daf9,  Localized=x: 1.4519350881997504, y: -31.02742566505306}

Where is this data coming from? That string for the `"Localized"` key is not a standard format that WPILib sends data in

Starlight220, 2024-03-18:
At minimum you're sending the field. What else are you sending?

Please go over your usage of the Shuffleboard API and check if you're publishing anything to the same path as the field.

Searching for "Localized" would also be a good guess to find what is being published and where.

feoranis26, 2024-03-19:
Across our entire SmartDashboard API calls, this line is the only one that includes the word "Localized", which seems suspiciously similar to the format of the data causing the exception. I don't think it's related to this issue though, as it publishes to a separate path. We also don't publish anything to the same path of the field.

` SmartDashboard.putString("Localized pose", "x: " + robot_actual_pose.getX() + ", y: " + robot_actual_pose.getY()); `

I'll look into our NT usage deeper once I'm able to work on the robot.

PR #798 (2024-04-04): Correct FieldData de/serialization
The first commit fixes #797
The second commit is a bug I saw and decided to fix on the way

--- 

The fix here is to explicitly check for the type being `double[]` and warning+dropping if not, rather than unsafely casting which leads to CCEs like #797 in cases where there's unrelated data in that table.
closed 2024-04-04 by PeterJohnson