wpilibsuite/shuffleboard #820: Updates to SimpleWidgets not seen by robot code.

edreed, 2025-01-11:
**Describe the bug**  
Updates to SimpleWidgets not seen by robot code.

Our robot code creates widgets that allow a human operator to change values in our code like PID parameters, RPM values, enabling features, etc. on the fly. With Shuffleboard 2025.1.1, updates to the SimpleWidget values are no longer being seen by the robot code. Updating the Network Table entries directly through Glass works as expected, i.e. these updates are seen by the robot code, so the problem seems to be specific the Shuffleboard.

Shuffleboard 2024.3.1 works as expected.

**To Reproduce**  
Steps to reproduce the behavior:
1. Run the following robot code in simulation:

  ```java
  package frc.robot;
  
  import edu.wpi.first.networktables.GenericEntry;
  import edu.wpi.first.wpilibj.TimedRobot;
  import edu.wpi.first.wpilibj.shuffleboard.BuiltInLayouts;
  import edu.wpi.first.wpilibj.shuffleboard.BuiltInWidgets;
  import edu.wpi.first.wpilibj.shuffleboard.Shuffleboard;
  import edu.wpi.first.wpilibj.shuffleboard.ShuffleboardLayout;
  import edu.wpi.first.wpilibj.shuffleboard.ShuffleboardTab;
  import edu.wpi.first.wpilibj.shuffleboard.SimpleWidget;
  
  public class Robot extends TimedRobot {
    private ShuffleboardTab tab = Shuffleboard.getTab("MyTab");
    private ShuffleboardLayout layout = tab.getLayout("MyLayout", BuiltInLayouts.kList)
      .withSize(2, 2);
    private SimpleWidget onOffButton = layout.add("On-Off", false)
      .withWidget(BuiltInWidgets.kToggleSwitch);
    private GenericEntry onOff = onOffButton.getEntry(); 
    private SimpleWidget onOffState = layout.add("State", "Off")
      .withWidget(BuiltInWidgets.kTextView);
    private GenericEntry state = onOffState.getEntry();
  
    public Robot() {
      onOff.getTopic().setRetained(true);
    }
  
    @Override
    public void robotPeriodic() {
      if (onOff.getBoolean(false)) {
        state.setString("On");
      } else {
        state.setString("Off");
      }
    }
  }
```
  
2. Start Shufffleboard 2025.1.1 and connect to localhost.
3. Navigate to the "MyTab" tab and toggle the value of the toggle switch.

**Expected behavior**  
The sample code should see the toggle switch changes and update the value of the "State" text box widget accordingly.

**Screenshots**  
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11 10.0 amd64
 - Java: JRE: 17.0.12+7 -- Eclipse Adoptium
 - Version: 2025.1.1 built on January 1, 2025

**Additional context**

edreed, 2025-01-11:
I'd be happy to debug and submit a fix for the issue, but I cannot build Shuffleboard given the instructions in the README. It fails with the following error. Any help with this would be greatly appreciated!

```
PS> ./gradlew :app:run

FAILURE: Build failed with an exception.
     
* What went wrong:
A problem occurred configuring root project 'shuffleboard'.
> Could not resolve all files for configuration ':classpath'.
   > Could not resolve edu.wpi.first:wpilib-tool-plugin:2.1.0.
     Required by:
         project : > edu.wpi.first.WpilibTools:edu.wpi.first.WpilibTools.gradle.plugin:2.1.0
      > No matching variant of edu.wpi.first:wpilib-tool-plugin:2.1.0 was found. The consumer was configured to find a library for use during runtime, compatible with Java 11, packaged as a jar, and its dependencies declared externally, as well as attribute 'org.gradle.plugin.api-version' with value '8.5' but:
          - Variant 'apiElements' capability edu.wpi.first:wpilib-tool-plugin:2.1.0 declares a library, packaged as a jar, and its dependencies declared externally:
              - Incompatible because this component declares a component for use during compile-time, compatible with Java 17 and the consumer needed a component for use during runtime, compatible with Java 11
              - Other compatible attribute:
                  - Doesn't say anything about org.gradle.plugin.api-version (required '8.5')
          - Variant 'javadocElements' capability edu.wpi.first:wpilib-tool-plugin:2.1.0 declares a component for use during runtime, and its dependencies declared externally:
              - Incompatible because this component declares documentation and the consumer needed a library
              - Other compatible attributes:
                  - Doesn't say anything about its target Java version (required compatibility with Java 11)
                  - Doesn't say anything about its elements (required them packaged as a jar)
                  - Doesn't say anything about org.gradle.plugin.api-version (required '8.5')
          - Variant 'runtimeElements' capability edu.wpi.first:wpilib-tool-plugin:2.1.0 declares a library for use during runtime, packaged as a jar, and its dependencies declared externally:
              - Incompatible because this component declares a component, compatible with Java 17 and the consumer needed a component, compatible with Java 11
              - Other compatible attribute:
                  - Doesn't say anything about org.gradle.plugin.api-version (required '8.5')
          - Variant 'sourcesElements' capability edu.wpi.first:wpilib-tool-plugin:2.1.0 declares a component for use during runtime, and its dependencies declared externally:
              - Incompatible because this component declares documentation and the consumer needed a library
              - Other compatible attributes:
                  - Doesn't say anything about its target Java version (required compatibility with Java 11)
                  - Doesn't say anything about its elements (required them packaged as a jar)
                  - Doesn't say anything about org.gradle.plugin.api-version (required '8.5')

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

For more on this, please refer to https://docs.gradle.org/8.5/userguide/command_line_interface.html#sec:command_line_warnings in the Gradle documentation.

BUILD FAILED in 1s
```

sciencewhiz, 2025-01-11:
I'm wondering if it somehow is picking up a different JDK. Can you post the output of `./gradlew -q javaToolchains`

sciencewhiz, 2025-01-11:
I confirmed you get that failure if gradle is using JDK 11. Make sure JAVA_HOME is pointing to JDK 17

sciencewhiz, 2025-01-11:
An even simpler repro is just `SmartDashboard.putBoolean("test", false);` in the Robot constructor and watch OutlineViewer or the SimGui when changing the value. It changes when using 2024 Shuffleboard, but not 2025 Shuffleboard

sciencewhiz, 2025-01-11:
bisecting shows that this was broken in #815. Before that, the publisher is created with a single leading slash. With that PR, the publisher is created with two leading slashes.

closed 2025-01-12 by PeterJohnson

edreed, 2025-01-12:
@sciencewhiz 

> I confirmed you get that failure if gradle is using JDK 11. Make sure JAVA_HOME is pointing to JDK 17

Interesting. `JAVA_HOME` was not set and `./gradlew -q javaToolchains` also failed with the same error. Once I set `JAVA_HOME` to the 2025 JDK, it worked just fine.

Thanks!

PR #821 (2025-01-12): Normalize DataSource key
Fixes #820
commit 88b7e76f (2025-01-12): Normalize DataSource key (#821)

Fixes #820