wpilibsuite/shuffleboard #823: Exception Access Violation 0xc000005 on windows build start

CaffeinatedOpe, 2025-01-18:
**Describe the bug**  
after building the latest version of the code from git, the application shows the loading popup and then crashes, leaving an exception access violation in the command prompt.

**To Reproduce**  
Steps to reproduce the behavior:
1. download latest code from github
2. use gradle to run the program, or build and manually run the program.

**Expected behavior**  
application launches and functions.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Java version: jdk 17
 - Version: latest from git

log file:
[hs_err_pid23020.log](https://github.com/user-attachments/files/18465697/hs_err_pid23020.log)

PeterJohnson, 2025-01-18:
This is probably due to picking up a system JDK instead of the WPILib installed one.  The native libraries we use require the latest runtime, which we bundle with our JDK install.

ThadHouse, 2025-01-18:
You are likely running with an older jdk that has a broken msvc runtime. Please use the wpilib installer, and run with the jdk and shortcuts shipped with that.

CaffeinatedOpe, 2025-01-18:
This build was using the version installed by the wpilib installer.

CaffeinatedOpe, 2025-01-18:
Yup, even just running the executable with the wpilib version of java works fine. Thank you guys

closed 2025-01-18 by CaffeinatedOpe